ncl-ROVers/arduino #25: Implement new communication schema

antoinepetty, 2020-01-24:
As agreed in #12

antoinepetty, 2020-01-25:
Also move serial.begin from setup function into communication constructor. May need object factory to ensure there's only one instance? not sure. Probably feature creep

antoinepetty, 2020-01-29:
Needs to send a message on boot, and maybe a keepalive with the ID when nothing has been received for a period of time?

TheCodeSummoner, 2020-02-08:
Raspberry Pi (on receive) expects 2-byte integer values (multiple bytes ofc), strips all special characters (`/n`, `/r` etc.). First 2 bytes are expected to be device ID. The other ones are processed according to some specialised functions.

Sends 2-byte integer values (multiple bytes ofc) finished with a `\n`. Order determined by dictionaries in `src/utils.py`, `raspberry-pi` repository.

No CRC needed on Pi side (parity checking built into serial lib).

closed 2020-02-23 by antoinepetty