ncl-ROVers/arduino #27: Create diagrams to show device connectivity

antoinepetty, 2020-02-03:


antoinepetty, 2020-02-08:
Need diagrams to show how all the ROV devices are connected to each other to assist with development

antoinepetty, 2020-02-17:
Output devices:

Array index | Description
-- | --
0 | Forward Port Thruster (front right)
1 | Forward Starboard Thruster (front left)
2 | Aft Port Thruster (back left)
3 | Aft Starboard Thruster (back right)
4 | Top Forward Port Thruster (front right)
5 | Top Forward Starboard Thruster (front left)
6 | Top Aft Port Thruster (back left)
7 | Top Aft Starboard Thruster (back right)
8 | Arm Gripper Motor

antoinepetty, 2020-02-23:
@TheCodeSummoner do you approve of the new shortened JSON IDs? (for faster transfer)

antoinepetty, 2020-02-23:
See https://github.com/ncl-ROVers/arduino/issues/34#issuecomment-590076953 for an example of new json keys

TheCodeSummoner, 2020-02-23:
@antoinepetty 

We're not using IMU, and we're not using the depth sensor.

Can you please include Arduino ID-s in this diagram, and per ID breakdown (so which has which output)?

Other than that, it looks ok to me.

TheCodeSummoner, 2020-02-23:
Actually, I'd switch the naming too -> output devices are more like input values - each Arduino will expect those to arrive from surface.

antoinepetty, 2020-02-23:
[image]

TheCodeSummoner, 2020-02-27:
Can we please use the following keys:

```
{
        "T_HFP": THRUSTER_IDLE,
        "T_HFS": THRUSTER_IDLE,
        "T_HAP": THRUSTER_IDLE,
        "T_HAS": THRUSTER_IDLE,
        "T_VFP": THRUSTER_IDLE,
        "T_VFS": THRUSTER_IDLE,
        "T_VAP": THRUSTER_IDLE,
        "T_VAS": THRUSTER_IDLE,
        "T_M": THRUSTER_IDLE,
        "MG": GRIPPER_IDLE,
        "MC": CORD_IDLE
    }
```

Also, I know `THRUSTER_IDLE` is 1500 - but what about the other values?

TheCodeSummoner, 2020-02-27:
Also, each device will need to send a value on start with the id -> `ARD_O` or `ARD_I`.

TheCodeSummoner, 2020-02-27:
Hhere's an updated working code sample for an Arduino:

```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("{\"id\": \"ARD_I\"}\n");
}

void loop() {
  // put your main code here, to run repeatedly:

}

/*
  SerialEvent occurs whenever a new data comes in the hardware serial RX. This
  routine is run between each time loop() runs, so using delay inside loop can
  delay response. Multiple bytes of data may be available.
*/
void serialEvent() {
  while (Serial.available()) {
    // get the new byte:
    char inChar = (char)Serial.read();
    // add it to the inputString:
    if (inChar == '\n' || inChar == '\r') {
      Serial.println("{\"id\": \"ARD_I\"}\n");
    }
  }
}
```

antoinepetty, 2020-02-27:
> Also, each device will need to send a value on start with the id -> `ARD_O` or `ARD_I`.

Can I request these are changed to either just e.g. `O` or `A_O` please - seems pointless to have 3 characters just to say it's an Arduino
(Maybe motor keys also need underscores after the M for consistency?)

Current behaviour on boot (Which seems to happen when the Pi code connects to the Arduino) is:
```
{"id":"o","so":"4"}
{"id":"o"}
{"id":"o","so":"0"}
{"id":"o","so":"-13"}
{"id":"o"}
{"id":"o","so":"1"}
```
(Example showing current Master branch, pre key changes #38 )

TheCodeSummoner, 2020-02-29:
`M_G`, `M_C`, `A_O` and `A_I` in that case.

THe example satisfies the requirement, but the keys are incorrect -> the new keys will have to be `A_O`, `so` is irrelevant (will result in an error if you send an unregistered on PI key).

antoinepetty, 2020-03-01:
@TheCodeSummoner The `so` key is shortened `Status_O` from last year. This is to pass meta information about what the code is doing (See Ret Codes section of readme).

A good example for this year is that you can tell if the code has finished booting. An issue with I2C communication which we will be using for Thrusters is that a loose electrical connection can cause the code to hang waiting for a response which will never come. So if you see that the current status is `4` for a long time then you know that something has gone wrong.

I would like to request keys `S_O` and `S_I` for this year

antoinepetty, 2020-03-01:
Oh I think I may have misread- You're saying that ID and Status keys have been merged? So The Pi knows the Arduino ID from the status key?

antoinepetty, 2020-03-01:
For this ticket to be complete, the readme should be updated to show all incoming and outgoing keys from each Arduino for complete transparency

closed 2020-03-08 by antoinepetty